espressif/esp-azure #86: iothub_client_sample_mqtt gets tls error and stops sending

tinfoilbroker, 2020-04-27:
The sample application wasn't actually how I noticed this. I'm getting this error in a project at work and was working backwards to close in on the problem. Now I've noticed that it also appears when only the sample is running.

E (802632) esp-tls: couldn't get hostname for :tinfoilbroker.azure-devices.net:
Error: Time:Mon Apr 27 15:56:35 2020 File:/home/tinfoilbroker/esp/esp-azure/azure-iot-sdk-c/iothub_client/src/iothubtransport_mqtt_common.c Func:InitializeConnection Line:2355 mqtt_client timed out waiting for CONNACK
Error: Time:Mon Apr 27 15:56:35 2020 File:/home/tinfoilbroker/esp/esp-azure/port/src/tlsio_esp_tls.c Func:tlsio_esp_tls_destroy Line:152 tlsio_esp_tls_destroy called while not in TLSIO_STATE_CLOSED.

shahpiyushv, 2020-06-04:
@tinfoilbroker , such error can occur if internet connectivity fails. Do you mean to say that the firmware does not recover from this situation and does not retry and continue?

shahpiyushv, 2020-07-02:
@tinfoilbroker , if this issue is still valid, can you tell how to reproduce? As asked previously, was there any issue with Internet conectivity?

tinfoilbroker, 2020-07-02:
@shahpiyushv I still don't know what the problem really was, but when I tried another network it didn't occur anymore. So from that point on I was pretty sure, that it had to do with my ISP/Router/Whatever. After switching my ISP the error never showed up as it did before. By "as it did before" I mean, as explained in the issue. I could however, even before switching ISPs, produce a tls error by turning off wifi. And in this case it would reconnect as expected. But when an error occured with the other ISP the reconnect never happend.
So... it might have something to do with internet connectivity, but I can't tell for sure. And reproduction probably won't work either, because it only ever happened like this with my older ISP/Router.