espressif/esp-azure #90: Help for implementing Azure WebSocket support

MartinTJDK, 2020-06-23:
Hi Espressif,
I know, that an issue regarding WebSocket is active, and has been for a long time.
I Espressif able to help regarding this "feature" request?
Thanks

MartinTJDK, 2020-07-02:
Based on ideas from https://github.com/espressif/esp-azure/issues/65 I have managed to get MQTT over WebSocket to work. The issue was related to missing SSL negotiation [mbedtls_ssl_conf_renegotiation(&tls->conf, MBEDTLS_SSL_RENEGOTIATION_ENABLED); in file /components/esp-tls/esp_tls_mbedtls.c]

shahpiyushv, 2020-07-02:
@MartinTJDK ,  the support for enabling this automatically when CONFIG_MBEDTLS_SSL_RENEGOTIATION=y will soon be available in ESP_IDF. Meanwhile, we have also updated azure-iot-c-sdk to the latest (pre)release LTS_06_2020_Ref01.

shahpiyushv, 2020-07-21:
@MartinTJDK , the esp-tls fix for re-negotiation is now available on esp-idf master and release/v4.0 branches.