espressif/esp-azure #92: Will there be an ESP32-S2 Azure coming soon?

askpatrickw, 2020-07-02:
Will there be an ESP32-S2 Azure coming soon?
Will it demonstrating using the Digitial Signature Peripheral for working with self-signed and Espressif signed boards?

shahpiyushv, 2020-07-03:
@askpatrickw I have already tested all 3 examples with ESP32-S2 and they work fine.
I will check on the demo for Digital Signature Peripheral, but that may happen some time later.

askpatrickw, 2020-07-03:
Thank you @shahpiyushv That's great to hear!

askpatrickw, 2020-07-04:
I should be more specific, what I (and I assume others) would like to see is the use of the Digital Signature Peripheral for securely managing the Azure IoT x.509 certificates. It would to good to also attach some documentation on if Espressif can per-cert the chips. I see it is a service offered by Espressif, but specifics related to Azure Iot alongside the sample code would be awesome.

Thanks so much!

askpatrickw, 2020-10-05:
@shahpiyushv  Are you aware of any issues in using these samples with IoT Central? Specifically you first need to connect with DPS as your hub is never known, then use the hub in the result.

closed 2024-07-16 by askpatrickw