espressif/esp-azure #96: Getting issues during  - clone the esp-azure project

zafarpatel7, 2020-07-26:
[image]

shahpiyushv, 2020-07-27:
@zafarpatel7 , I think just trying again should fix the issue. esp-azure has a single submodule reference (azure-iot-sdk-c). All other submodules are part azure-iot-sdk-c, and not in our control. azure-iot-sdk-c is a huge repo with many submodules and so, a network error can cause errors during cloning.

zafarpatel7, 2020-07-27:
> @zafarpatel7 , I think just trying again should fix the issue. esp-azure has a single submodule reference (azure-iot-sdk-c). All other submodules are part azure-iot-sdk-c, and not in our control. [...]

[image]
 it's working ---Thnx

Noumeer, 2022-08-15:
I am facing the same issue and its not solving by cloning again, does there is any other solution to it?