espressif/esp-azure #98: undefined reference to `IoTHubClient_CreateFromConnectionString

schroedermarius, 2020-08-04:
I cant use the "high-level" api functions, only the Low-Level (_LL_) functions are recognized.
If I'm switchting to the normal functions I'm getting the following errors. Am I missing out on something?

```
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_client.cpp.obj):(.literal._ZN17AzureIotHubClient17azure_iothub_initEv+0x74): undefined reference to `IoTHubClient_CreateFromConnectionString'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_client.cpp.obj):(.literal._ZN17AzureIotHubClient17azure_iothub_initEv+0x78): undefined reference to `IoTHubDeviceClient_SetOption'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_client.cpp.obj):(.literal._ZN17AzureIotHubClient17azure_iothub_initEv+0x7c): undefined reference to `IoTHubClient_SetRetryPolicy'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_client.cpp.obj):(.literal._ZN17AzureIotHubClient17azure_iothub_initEv+0x80): undefined reference to `IoTHubDeviceClient_SetConnectionStatusCallback'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_client.cpp.obj):(.literal._ZN17AzureIotHubClient17azure_iothub_initEv+0x84): undefined reference to `IoTHubDeviceClient_SetMessageCallback'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_client.cpp.obj): in function `AzureIotHubClient::azure_iothub_init()':
/Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure_iothub_client.cpp:260: undefined reference to `IoTHubClient_CreateFromConnectionString'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure_iothub_client.cpp:267: undefined reference to `IoTHubDeviceClient_SetOption'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure_iothub_client.cpp:280: undefined reference to `IoTHubDeviceClient_SetOption'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure_iothub_client.cpp:280: undefined reference to `IoTHubDeviceClient_SetOption'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure_iothub_client.cpp:281: undefined reference to `IoTHubClient_SetRetryPolicy'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure_iothub_client.cpp:298: undefined reference to `IoTHubDeviceClient_SetOption'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure_iothub_client.cpp:298: undefined reference to `IoTHubDeviceClient_SetConnectionStatusCallback'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure_iothub_client.cpp:298: undefined reference to `IoTHubDeviceClient_SetMessageCallback'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_messaging.cpp.obj):(.literal._ZN20AzureIotHubMessaging31azure_iothub_messaging_generateEiPc+0x3c): undefined reference to `IoTHubDeviceClient_SendEventAsync'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_messaging.cpp.obj): in function `AzureIotHubMessaging::azure_iothub_messaging_generate(int, char*)':
/Users/marius/esp/esp-azure/examples/MyProject/build/../main/azure/messaging/azure_iothub_messaging.cpp:82: undefined reference to `IoTHubDeviceClient_SendEventAsync'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_device_twin.cpp.obj):(.literal._ZN21AzureIotHubDeviceTwin31azure_iothub_device_twin_updateEv+0x24): undefined reference to `IoTHubDeviceClient_SendReportedState'
/Users/marius/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(azure_iothub_device_twin.cpp.obj):(.literal._ZN21AzureIotHubDeviceTwin29azure_iothub_device_twin_initEv+0x4): undefined reference to `IoTHubDeviceClient_SetDeviceTwinCallback'
```

Darkenkade42, 2020-08-28:
It works as intended. You can only use "high-level" api functions if you provide your own  as specified in [thread api and sleep](https://github.com/Azure/azure-c-shared-utility/blob/master/devdoc/threadapi_and_sleep_requirements.md). 
You would also have to modify CMakeLists.txt to take proper files containiner "high-level" api functions and remove the current FreeRTOS thread api and sleep implementation (which does basically nothing and will give you error message if you try to use high-level functions without fixing it).

schroedermarius, 2020-09-07:
@Darkenkade42 Thanks for your response. Is there any sample where I could look at? Im quite new to this. Thanks in advance.

Darkenkade42, 2020-09-27:
@schroedermarius  This [fork](https://github.com/fgiancane8/esp-azure/tree/master/).