espressif/esp-azure #99: Error from SSL_write: -26752

durgeshIISc, 2020-08-10:
# Environment
Development Kit: ESP32-DevKitC
Kit version (for WroverKit/PicoKit/DevKitC): v1
Module or chip used: ESP32-WROOM-32D
IDF version (run git describe --tags to find it):
v4.0.1
esp-azure commit hash: b67efd7

Build System: idf.py
Compiler version (run xtensa-esp32-elf-gcc --version to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
Operating System: macOS
Using an IDE?: VSCode
Power Supply: USB

# Problem Description

I am connecting to my IoT hub using connection string. I am sending some burst of data to my IoT Hub. after each burst there is a wait time of 4 minutes. after sending the first burst of data  and completing the waiting time some times I get a SSL write error of -26752 and then a there are two things that happen
- The error goes away after some time and data is received on the IoT Hub.
- The error leads to mqtt failed to publish error message. and i get a Core panic message.

See the attached file for more detail.
LOG FILE: 
[mcu_log_N_2000_2_2_2.log](https://github.com/espressif/esp-azure/files/5049353/mcu_log_N_2000_2_2_2.log)

# Questions
 - What does the SSL write error code signify?
- what might be the possible reasons for the error.

Eavo, 2020-08-19:
1. Error code is MBEDTLS_ERR_SSL_WANT_WRITESSL.

2. It general means Write Timeout, The possible reason is the network is in poor condition.

3. You can call IoTHubClient_LL_SetRetryPolicy function for enable the IoT hub retry again.

teekaytk, 2020-10-21:
I have similar issue. I see once the Info: Error from SSL_write: -26752 appears , the sdk still accepts new messages until the heap is consumed and the device software reboots.
May be there is an elegant way detecting a missing network so the Azure SDK can notify the messaging thread to stop while it is waiting for connection but the detection seems really slow or broken so if you are sending a lot of messages very quickly it is possible to overwhelm the system.

durgeshIISc, 2020-10-23:
My issue is not solved. If you have some way of solving this issue please share.