espressif/esp-azure #100: *** SAS TOKEN EXPIRE RECEIVED ****

zafarpatel7, 2020-08-15:
Hi 

I am facing *** SAS TOKEN EXPIRE RECEIVED **** issues -- 

when I power On my board it is connected successfully but after some time (5-6 days)  it disconnects automatically and not connected until I do not restart the board, I did not understand the issues exactly -- Azure issue or FW issues, 

Could you please help me 

[image]

zafarpatel7, 2020-08-18:
any Suggestion on that -- 
- SAS TOKEN EXPIRE TIME set by ESP32 or Azure
- What is the default value of SAS TOKEN EXPIRE TIME
- esp-azure sdk will take care of SAS TOKEN EXPIRE time or we have to take care at the application FW side

Eavo, 2020-08-19:
1. Right

2. the default value of SAS TOKEN EXPIRE TIME is 3600, you can find it by DEFAULT_SAS_TOKEN_EXPIRY_TIME_SECS. You also can change it by IoTHubClient_Auth_Set_SasToken_Expiry and get it by IoTHubClient_Auth_Get_SasToken_Expiry.

3. It's better to take care of SAS TOKEN EXPIRE TIME in your FW side.

zafarpatel7, 2020-09-13:
Yes, as per your suggestion we found all this details. But in our project we are using "IoTHubClient_LL_CreateFromConnectionString" API to get the azure handle. So, what we saw your provided API required other IoTHub_auth handle. I have few doubts regarding same.
1. Do we need to change the firmware implementation to get the auth handle then we can able to change the SAS TOKEN EXPIRE time.
2. Is there any way on azure cloud side using that we can able to set or refresh the SAS TOKEN EXPIRE time. If we can use our current implementation.
3. Or is it ok to take both auth and client ll types of azure handle at firmware side and through that we can able to control it.

zafarpatel7, 2020-09-28:
anyone have 

> Yes, as per your suggestion we found all this details. But in our project we are using "IoTHubClient_LL_CreateFromConnectionString" API to get the azure handle. So, what we saw your provided API [...]

zafarpatel7, 2020-09-28:
> anyone have

Does anyone have any soution ????

Darkenkade42, 2020-09-30:
Might want to take a look at this document [Iothub SDK Options](https://github.com/Azure/azure-iot-sdk-c/blob/2019-03-18/doc/Iothub_sdk_options.md). Function you're looking to use is IoTHubDeviceClient_LL_SetOption. Do you stop calling Do_Work after disconnect?