espressif/esp-azure #101: Cloning Issue

intothefuture20, 2020-08-18:
While cloning got the following errors:

Failed to recurse into submodule path 'azure-iot-sdk-c/provisioning_client/deps/utpm'
Failed to recurse into submodule path 'azure-iot-sdk-c/uamqp'
Failed to recurse into submodule path 'azure-iot-sdk-c'

[image]

Seeking assistance please.

shahpiyushv, 2020-08-18:
@intothefuture20 I think just trying again should resolve this issue. ESP Azure just has a single sub-module azure-iot-sdk-c, but azure-iot-sdk-c itself has many more sub-modules.

intothefuture20, 2020-08-18:
Hello Piyush, I tried again and its same result... Same Error

shahpiyushv, 2020-08-18:
@intothefuture20 , that is strange. I just tried recursive cloning using command line as well as github desktop app. Both worked fine. Seems there could be some network issue on your side?

If the parent esp-azure repo has been cloned already, can you just try fetching the submodules a few more times using the below from esp-azure folder?

```
git submodule update --init --recursive
```

zafarpatel7, 2020-08-18:
HI @intothefuture20 
if your c/d driver path is long it will not work - so direct Clone at C: or D: (D:\esp-azure) then it will work

intothefuture20, 2020-08-19:
Hello Zafar,

Well it did work when I cloned it on root like c drive or D drive. Thank you all for the help. Much Appreciated!

intothefuture20, 2020-08-19:
The issue is resolved. Closing the issue. Thank you!

closed 2020-08-19 by intothefuture20