espressif/esp-azure #107: Any plan to add PnP support by updating azure iot c sdk to LTS_07_2020_Ref01?

xiongyu0523, 2020-12-07:
We have some partners want to certify their ESP based device to Azure PnP. They start from here so I'm checking whether we can update SDK to the latest LTS. Thanks.

shahpiyushv, 2020-12-07:
@xiongyu0523 , as of now, we have PNP support being maintained on a separate branch [preview/pnp_example](https://github.com/espressif/esp-azure/tree/preview/pnp_example). Have you checked that out?

xiongyu0523, 2020-12-08:
that's great. I will check it.

xiongyu0523, 2020-12-08:
It works perfectly on my ESP32 board. Thanks.

In PnP certification requirement document, it explicitly ask for IDscope transfer capability. Can ESP32 platform support this?

```
Device must implement easy transfer of DPS ID Scope ownership without needing to recompile the embedded code
```

lhespress, 2020-12-09:
@xiongyu0523 ESP32 platform has support DPS ID Scope,  but it need to recompile the embedded code. we haven't provide without recompile the embedded code.

xiongyu0523, 2020-12-10:
I understand your IDscope is in Kconfig. That might be treat as a 'no code' solution. let me check with people who take care of certification. THanks