espressif/esp-azure #113: Proper way to use esp-azure inside an already existing project

Riborn, 2021-02-19:
Hi,

I'm new to ESP-IDF and I would like to use the esp-azure SDK within an already existing  ESP-IDF project using Eclipse IDE.
I already installed ESP-IDF Toolchain and added the Espressif plugin to Eclipse.
May be it sounds like a dumb question, but what I am supposed to do once I have cloned the ESP Azure SDK on my computer ?
I would like to find a way to combine the examples given (mainly "iothub_client_sample_mqtt") with my own project, but I can't find a step by step explanation about the way to set everything up.

Thanks for your help !

liudr, 2021-03-11:
#114 has my method.

Riborn, 2021-03-12:
> #114 has my method.

Thanks a lot! Adding the absolute path of esp-azure/port in my project root CMakeLists.txt with set(EXTRA_COMPONENT_DIRS "C:/path/to/esp-azure/port") worked for me

lhespress, 2021-03-15:
@Riborn If the issue has been resolved, please help close this issue. Thanks.

closed 2021-03-15 by Riborn