espressif/esp-azure #114: Example project "iothub_client_sample_mqtt" does not compile in Eclipse

freeride01, 2021-02-25:
Hi,
I've been using Eclipse with ESP-IDF for a while now, and now I want to use esp-azure to connect to the Azure IoT Hub.
I followed the guide and cloned esp-azure in another directory on my hard drive (ESP-IDF 4.0 is in C:\sdks\esp-idf, esp-azure is in C:\sdks\esp-azure).
However, the compiler cannot find the included files in iothub_client_sample_mqtt.c. First it could not find iothub_client.h, I put the absolute file path and now it finds it, but now it can't find umock_c/umock_c_prod.h (the first included file in iothub_client.h). I don't want to have to manually add the absolute path to each and every included file, furthermore for some files there are multiple options and I don't know which is the one to include.
What is the proper way to make the example projects work in Eclipse and is this even supported at this time?
Thanks in advance.

liudr, 2021-03-11:
They have a nasty relative path here in CMakeLists.txt:

set (EXTRA_COMPONENT_DIRS "${CMAKE_CURRENT_BINARY_DIR}/../../../port")

Changing this one into an absolute path should solve all your problems.
What I've been doing is to define an AZURE_PATH for the absolute path so it would read $ENV{AZURE_PATH}/port). You can place this define in the same place where you find IDF_PATH in eclipse.

Riborn, 2021-03-12:
Hi, I had the same problem. Adding the absolute path of esp-azure/port in my project root CMakeLists.txt with set(EXTRA_COMPONENT_DIRS "C:/path/to/esp-azure/port") worked for me

freeride01, 2021-03-12:
Thank you both very much! The project works now.