espressif/esp-azure #115: Dependency on main prevents use from other components

NathanJPhillips, 2021-03-10:
Referencing `main` from CMakeLists.txt creates a circular dependency that will in many cases cause undefined references during linking when esp-azure is `REQUIRE`d by a component that is in turn `REQUIRE`d by the main project.

Is this a known issue, are there reasons `main` needs to be referenced, or should I create a PR to remove this?

liudr, 2021-03-11:
I'm curious, what is the reason that main is required in your CMakeLists.txt? I'm learning the CMake build system. This might help me understand what I don't know that I don't know. ;)

jspngh, 2021-07-16:
I ran into the circular dependency issue as well after restructuring some code.
I also don't see a reason for the dependency on `main`, so it would be good if someone from Espressif can comment whether it can be removed or not.