espressif/esp-azure #116: Any plan to rename this "port" folder?

liudr, 2021-03-11:
Due to the way ESP-IDF components work, the azure as an ESP-IDF component is given a name "port" because the folder is, simply put "port", inside azure folder such as ".espressif\esp-azure\port". Is there any plan to rename this something like "esp_azure_port" to make more sense? It's very quiet here. I used to ask questions and now I'm answering here. Where are the developers? :)

lhespress, 2021-03-12:
@liudr There isn't any plan to rename this "port" folder now. it keeps the same struct with the other components.

liudr, 2021-03-14:
Then is there an elegant way to incorporate esp-azure as a component in other components with descriptive names like esp_netif or esp-tls? Which other components have been using a name as generic as "port"? They would be overriding one another if they share name wouldn't they?

lhespress, 2021-03-15:
@liudr There is the other components, such as `esp-freertos-core`, `esp-google-iot`, `esp-aws-iot` and so on. which have been using a name as "port". It's better to incorporate esp-azure as a independently component. It need modify CMakefile if you want to add it to the other components.