espressif/esp-azure #122: Update to LTS_07_2021_Ref01

MartinTJDK, 2021-11-18:
Any plans for update to LTS_07_2021_Ref01?

MartinTJDK, 2021-12-14:
@mahavirj Do you know, if it is prioritised?

lhespress, 2021-12-15:
@MartinTJDK There isn't plan for update to LTS_07_2021_Ref01? What new functions for your project need update it?

MartinTJDK, 2021-12-15:
@lhespress LTS_07_2021_Ref01 has several bugfixes, which we need.
When continiously removing network connectivity, the "esp-azure" in combination with ESP network driver locks up, and never gets online again. Even resolving of host names fails at that point.
I have also seen 0-pointer issues, which should have been fixed in azure sdk.
All mentioned issues are only seen, when network is not stable.

epictrave, 2021-12-30:
@mahavirj @MartinTJDK @lhespress I think this repo needs LTS_07_2021_Ref01.
Because of upcoming TLS certificate changes (from the current Baltimore CyberTrust CA Root to the DigiCert Global G2 CA root)

MartinTJDK, 2022-01-04:
@epictrave I agree. The certificate bundle has to be updated. But it would also be nice to get bugfixes as well.

MartinTJDK, 2022-01-12:
@mahavirj @lhespress Is it something you will consider?

lhespress, 2022-01-18:
@MartinTJDK As mentioned earlier, no updates are planned in the near future, we are discussing which to update because of there are both [Azure-iot-SDK-C](https://github.com/Azure/azure-iot-sdk-c) and [Azure-SDK-for-c](https://github.com/Azure/azure-sdk-for-c) to access Azure-IoT.

matgardon, 2022-02-14:
@lhespress I'm not sure this has been stated clearly enough by @epictrave so I'll just stress it again here: without this LTS, the source code will not contain the right CA for ESP32 devices to be able to connect to azure anymore, as explained [in this blog post](https://techcommunity.microsoft.com/t5/internet-of-things-blog/azure-iot-tls-critical-changes-are-almost-here-and-why-you/ba-p/2393169).

So getting this update, if not for all the other stability fixes, doesn't seem like optional for people using this repo when building their binaries for ESP devices connecting to azure.

If it's not officially upgraded, we will have to upgrade ourselves one way or another, or devices will not be able to connect to azure anymore (and that will happen soon).

lhespress, 2022-02-17:
@matgardon We have create a merge request for update it to [LTS_01_2022_Ref01](https://github.com/Azure/azure-iot-sdk-c/releases/tag/LTS_01_2022_Ref01) and will push it after review finish.

matgardon, 2022-02-18:
@lhespress awesome news, thanks for the follow up :pray:

shahpiyushv, 2022-02-18:
This has now been pushed out to the repository.