espressif/esp-azure #126: Not able to auto-reconnect with Azure Cloud

harshalbhav, 2022-04-17:
## Environment
- Development Kit: ESP32-Wrover-Kit and Other
- Kit version (for WroverKit/PicoKit/DevKitC):
- Module or chip used: ESP32-WROOM-32D
- IDF version (run git describe --tags to find it):
- // v4.4
- Build System: idf.py
- Compiler version (run xtensa-esp32-elf-gcc --version to find it):
- // 1.22.0-80-g6c4433a
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply: USB and external 3.3

## Problem Description

We have developed one hardware based on ESP32. We are using Azue-SDK for same development. We enabled debug flag for Azure-SDK so we are receiving all debug logs on console.

Right now we are facing issue with Azure-SDK. We have received disconnect event from Azure stack. As per the Azure stack we are expecting following points in this scenario.

Azure cloud should be update the device status like Device is disconnected.
Device should be reconnect to the Azure cloud automatically.
We are not able to full fill above both points. So, our Device and Azure cloud not able to get into valid state.

Please update us how to drive this thing.

Thanks,
Harshal

## Expected Behavior
Azure cloud should be update the device status like Device is disconnected.
Device should be reconnect to the Azure cloud automatically.
Confirmation[346] received for message tracking id = 346 with result = IOTHUB_CLIENT_CONFIRMATION_OK
-> 06:26:46 DISCONNECT
Connection Status result: IOTHUB_CLIENT_CONNECTION_UNAUTHENTICATED, Connection Status reason: IOTHUB_CLIENT_CONNECTION_EXPIRED_SAS_TOKEN

******* SAS TOKEN EXPIRE RECEIVED **********

-> 06:26:49 CONNECT | VER: 4 | KEEPALIVE: 240 | FLAGS: 192 | USERNAME: .... | PWD: XXXX | CLEAN: 0

<- 06:26:49 CONNACK | SESSION_PRESENT: true | RETURN_CODE: 0x0

## Actual Behavior
Azure cloud not getting Device status as its disconnected.
Device is not able to reconnect with Azure cloud. (We had wait till 5 to 8 min)
Confirmation[346] received for message tracking id = 346 with result = IOTHUB_CLIENT_CONFIRMATION_OK
-> 06:26:46 DISCONNECT
Connection Status result: IOTHUB_CLIENT_CONNECTION_UNAUTHENTICATED, Connection Status reason: IOTHUB_CLIENT_CONNECTION_EXPIRED_SAS_TOKEN

******* SAS TOKEN EXPIRE RECEIVED **********
.
.
.

## Steps to reproduce
No any particular step we observed.

zafarpatel7, 2022-04-22:
Hello @harshalbhav 

I am also facing the same issue, in 24 hr 3-5 times I am getting the same issues, once I get this kind of error I am doing soft-reset which is not good looks to me

Hello Team Espressif Systems

Could you please help us with this issue

zafarpatel7, 2022-04-27:
@mahavirj could u plz help me on this

lhespress, 2022-05-07:
@zafarpatel7 Please provide the example when you face the issue firstly and then check that the work thread is  exit. such as : ```iothub_client_sample_mqtt has gotten quit message``` show on https://github.com/espressif/esp-azure/blob/master/examples/iothub_client_sample_mqtt/main/iothub_client_sample_mqtt.c#L229

harshalbhav, 2022-05-09:
@lhespress Thanks for update. We are not receiving any thread exit message as you mentioned into the link. Also we have take care DoWork function call. In my case all debug logs are enabled. 

We had used same sample code [**iothub_client_sample_mqtt**].

Thanks,
Harshal

zafarpatel7, 2022-06-25:
Hello All 
can anyone help us on this issue?

@mahavirj @shahpiyushv @wujiangang @jitin17