espressif/esp-azure #127: Loss of wifi AP connection results in Azure/TLS errors on recovery (IDFGH-7210)

rtheil-growlink, 2022-04-22:


hauserkristof, 2022-05-27:
I have the exact same issue! Could somebody managed to avoid it?

rtheil-growlink, 2022-05-27:
I switched to the Azure middleware for FreeRTOS. Worth the effort. https://github.com/Azure/azure-iot-middleware-freertos

Vivero, 2022-10-07:
I was running into this same exact issue. Sometimes I'd get into this scenario even without turning off the AP. My device would just randomly disconnect and then never recover. The only workaround I was able to find was to modify a part of the MQTT code myself. Here's what that fix looks like (using my own fork of `azure-umqtt-c`): https://github.com/MvIt-Now/azure-umqtt-c/commit/bcb0b2387fd04cc32c2d53d4d65363c296188af8

afcec, 2022-12-07:
I have also the same issue but unfortunately this repo seems to be dead...

rtheil-growlink, 2022-12-07:
@afcec Look into Azure middleware. They have a samples repo that has everything you need for getting up and running on any espressif device.

ClaesIvarsson, 2023-09-29:
I just found a problem in tlsio_esp_tls.c
When a problem occur ion the socket, the state goes to TLSIO_STATE_ERROR, however in most cases it is not reported upstream, which means azure sdk is not aware of the problem.
My solution was to change the code to always call enter_tlsio_error_state when current code set the error state
Happens only in the function tlsio_esp_tls_dowork:

`
static void tlsio_esp_tls_dowork(CONCRETE_IO_HANDLE tls_io)
{
    if (tls_io == NULL)
    {
        /* Codes_SRS_TLSIO_30_070: [ If the tlsio_handle parameter is NULL, tlsio_dowork shall do nothing except log an error. ]*/
        LogError("NULL tlsio");
    }
    else
    {
        TLS_IO_INSTANCE* tls_io_instance = (TLS_IO_INSTANCE*)tls_io;

        // This switch statement handles all of the state transitions during the opening process
        switch (tls_io_instance->tlsio_state)
        {
        case TLSIO_STATE_CLOSED:
            /* Codes_SRS_TLSIO_30_075: [ If the adapter is in TLSIO_STATE_EXT_CLOSED then  tlsio_dowork  shall do nothing. ]*/
            // Waiting to be opened, nothing to do
            break;
        case TLSIO_STATE_INIT:
            {
                int result = esp_tls_conn_new_async(tls_io_instance->hostname, strlen(tls_io_instance->hostname), tls_io_instance->port, &tls_io_instance->esp_tls_cfg, tls_io_instance->esp_tls_handle);
                if (result == 1) {
                    tls_io_instance->tlsio_state = TLSIO_STATE_OPEN;
                    tls_io_instance->on_open_complete(tls_io_instance->on_open_complete_context, IO_OPEN_OK);
                } else if (result == -1) {
                    enter_tlsio_error_state(tls_io_instance);
                }
            }
            break;
        case TLSIO_STATE_OPEN:
            if (dowork_read(tls_io_instance) < 0 && errno != EAGAIN)
            {
                enter_tlsio_error_state(tls_io_instance);
            }
            if (dowork_send(tls_io_instance) < 0 && errno != EAGAIN)
            {
                enter_tlsio_error_state(tls_io_instance);
            }
            break;
        case TLSIO_STATE_ERROR:
            /* Codes_SRS_TLSIO_30_071: [ If the adapter is in TLSIO_STATE_EXT_ERROR then tlsio_dowork shall do nothing. ]*/
            // There's nothing valid to do here but wait to be retried
            break;
        default:
            LogError("Unexpected internal tlsio state");
            break;
        }
    }
}

`