espressif/esp-azure #134: tls_error when compiling  esp-idf 5

preveen-stack, 2023-05-22:
I was tying to compile the sample application iothub_client_sample_mqtt with esp-idf 5.0.0 and is getting the below errors. Has anybody come across them? 
```
esp5/esp-idf/esp-azure/port/src/tlsio_esp_tls.c:127:5: error: implicit declaration of function 'esp_tls_conn_delete'; did you mean 'esp_tls_conn_write'? [-Werror=implicit-function-declaration]
  127 |     esp_tls_conn_delete(tls_io_instance->esp_tls_handle);
      |     ^~~~~~~~~~~~~~~~~~~
      |     esp_tls_conn_write
```
```
/esp5/esp-idf/esp-azure/port/src/tlsio_esp_tls.c: In function 'tlsio_esp_tls_create':
/Users/preveen/esp5/esp-idf/esp-azure/port/src/tlsio_esp_tls.c:219:59: error: invalid application of 'sizeo' to incomplete type 'esp_tls_t' {aka 'struct esp_tls'}
  219 |                 result->esp_tls_handle = calloc(1, sizeof(esp_tls_t));
      |                                                           ^~~~~~~~~
```

dhammuabhi, 2023-05-26:
Yes I too came accross this issue. Can anyone help me to solve this issue

Rainbow-Cai, 2023-05-30:
Hi  There are no alternative functions for the "esp_tls_conn_delete()"  in the 5.0 SDK. Please refer to the [“Function Deprecations And Recommended Alternatives” ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/protocols.html#function-deprecations-and-recommended-alternatives)programming guide.
[image]

dhammuabhi, 2023-06-07:
second issue still didn't resolve 

/home/ubuntu22/esp/esp-azure/port/src/tlsio_esp_tls.c:220:59: error: invalid application of 'sizeof' to incomplete type 'esp_tls_t' {aka 'struct esp_tls'}
  220 |                 result->esp_tls_handle = calloc(1, sizeof(esp_tls_t));
      |

someone42, 2023-08-11:
To anyone having this issue, the second error can be resolved by replacing the line:
`result->esp_tls_handle = calloc(1, sizeof(esp_tls_t));`
with:
`result->esp_tls_handle = esp_tls_init();`