espressif/esp-azure #138: error: implicit declaration of function 'esp_tls_conn_delete'

anoopkumares, 2023-11-02:
I am trying to compile iothub_client_sample_mqtt example. I have cloned esp-idf repo and this repo. but while compiling the code I am getting the following error:

`tlsio_esp_tls.c:127:5: error: implicit declaration of function 'esp_tls_conn_delete'; did you mean 'esp_tls_conn_write'? [-Werror=implicit-function-declaration]
  127 |     esp_tls_conn_delete(tls_io_instance->esp_tls_handle);`

How can I fix this error ?

aleblsv, 2023-11-06:
@anoopkumares 
This is because you compile with esp-idf v5
Change from esp_tls_conn_delete to esp_tls_conn_destroy

anoopkumares, 2023-11-07:
I fixed that. But I am getting another error.
`tlsio_esp_tls.c:219:59: error: invalid application of 'sizeof' to incomplete type 'esp_tls_t' {aka 'struct esp_tls'}
  219 |                 result->esp_tls_handle = calloc(1, sizeof(esp_tls_t));
      |                                                           ^~~~~~~~~
[14/84] Building C object esp-idf/port...iot-sdk-c/c-utility/src/sastoken.c.obj
ninja: build stopped: subcommand failed.
`

aleblsv, 2023-11-07:
@anoopkumares 
Just try to allocate fixed memory, in my case seems that 2*1024 works as expected.
 
                // result->esp_tls_handle = (esp_tls_t*)calloc(1, sizeof(esp_tls_t));
                //ToDo: esp-idf v5.1.1 compilation problem of un-complete type sizeof(esp_tls_t), below is a workaround
                result->esp_tls_handle = (esp_tls_t*)calloc(1, 2*1024);